tomcl/issie #329: Unable to install Issie newest release on M1 Mac

bryan-terra, 2023-07-21:
**Describe the bug**
I am unable to install the newest release of Issie on my M1 Mac.

The installer is called issie-4.0.2-arm64.dmg but the release number is 4.0.a.3, could the issue be that the installer is out of date?

https://github.com/tomcl/issie/releases/tag/v4.0.a.3

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/tomcl/issie/releases/tag/v4.0.a.3
2. Download issie-4.0.2-arm64.dmg 
3. Open the installer and complete installation
4. Open the app by clicking on icon and get the error: 

"issie” is damaged and can’t be opened. You should move it to the Bin."

**Expected behavior**
Issie should launch correctly

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Version: 13.4.1 (22F82)
 - Apple M1 Processor

0x6770, 2023-07-22:
I've updated the Mac installer to v4.0.a.3. Please download the latest installer and check if the issue still exists. Also, ensure your Mac's security setting allows apps downloaded from "App Store and identified developers."

[image]

tomcl, 2023-07-22:
When I create new releases on the alpha code I create windows only. @0x6770 can create silicon mac releases - he has a silicon mac. Anyone with a non-silicon mac can create x86 mac releases.

For those developing Issie the releases don't matter. Just `npm run dev` on master and you get the latest code 9which will often be better than the latest release. Then, on your architecture, you can try `npm run dist`. I am not sure whether this works simply on non-windows architectures at the moment, but Yukie will know. So you can generate your own binaries if you want that.

bryan-terra, 2023-07-22:
Hi all, many thanks for the quick response. 

I confirmed the security settings and tried the new installer, but the issue persists.

Does this info help at all?
[image]

tomcl, 2023-07-22:
It looks like you need to follow the override security settings link and do that?

tomcl, 2023-07-23:
PS - you will need to wait for 4.0.a.4 to have  apple mac binaries since the previous two versions had a serious error, now fixed.

tomcl, 2023-08-30:
Theo - please close this when all needed info from it is on wiki!

closed 2023-09-12 by tomcl