tomcl/issie #332: Improve Waveform Simulator

tomcl, 2023-07-24:
This issue has three parts, which need not all be done together.

1. Speed up waveform SVG generation by generating only the displayed part of the waveforms (normally approx 1/40 of what is now generated). These are now the main bottleneck in response for Issie. See generateWaveform in [wavesim.fs](https://github.com/tomcl/issie/blob/982c0337d68ad4df25451500be54bd1e9a6e1c9e/src/Renderer/UI/WaveSim/WaveSim.fs#L142).
2. Improve waveform numeric display when value constant across much or all of screen. The current display leads to weird spacing of numbers in some cases making it difficult to visualise waveform.
3. Alter the cursor and screen horizontal movement UI. Currently it is not quite right. Suggest:
    * single arrows move cursor by 1 as now on waveforms moving waveforms as well when cursor is with 20% of edges. (Currently it goes all the way to the screen edges)
    * double arrows move cursor relative to waveforms as now. They do NOT move cursor relative to window edge but move waveforms 50% of screen width keeping cursor position fixed.
    * Allow left/right arrows to move cursor
    * Changing cursor cycle number moves cursor and waveforms consistently with double arrow, e.g. cursor stays in same place if possible and waveforms move: if too near an edge then cursor position on screen changes
    * Special care is needed when the number of displayed cycles is very small (maybe then cursor can go all the way to the edge).

lucasng32, 2023-08-04:
Could I be assigned to this?

tomcl, 2023-09-25:
This will not be done Summer 2023. Leave it open as ideas - but by the time it is implemented other things will have changed.

tomcl, 2024-03-29:
https://github.com/tomcl/issie/issues/442#issuecomment-2027207069

This bug motivates doing the waveform generation properly – generating only the visible part. That then allows software  horizontal scrolling.

Two methods:

*All-F#-software*. Recreate waveforms with new horizontal position based on current scroll position. Quantise scrolling so that waveforms move an integral number of clock cycles. 
•	Scroll bar slider will snap back to the correct position (based on actual waveform scroll) when drag ends. 
•	This requires on-screen SVG generation and display to be < 20ms or so.

*Software-HTML* Recreate waveform SVGs with some not visible horizontal margins. Use HTML scrolling – this can therefore be continuous with no performance penalty. Or, it could be restricted to an integral number of clock cycles. Recreate waveforms when scrolling requires this, adding new margins. 
•	This method will require some fiddly pixel-based calculations to know exactly where to place the SVG, and which clock cycles to generate. 
•	This method means making the current top line display with cycle numbers move with the waveforms.
•	This requires on-screen (+ margin) SVG generation to be < 200ms or so.

**Vertical scrolling**

We could keep this as now, and limit number of waveforms so there is not a large number. Or, keep html vertical scrolling. Work out obviously off-screen waveforms. Delete them. Note that we don’t need to delete the text SVG – just remove it from the rendered DOM so it does not cause memory problems there.

samuelpswang, 2024-08-05:
Goals for this issue discussed in-depth in #448. Features implemented in PR #456.

closed 2024-08-05 by samuelpswang