tomcl/issie #383: smart routing bug from MUX

tomcl, 2023-08-30:
**Describe the bug**
in some circumstances smart routing goes wrong. This bug affects production designs, e.g. demo/eep1

**To Reproduce**
Steps to reproduce the behavior:
1. Load project `./demo/testrouting` on Issie branch `dev-testrouting`
2. Open sheet
3. compare the autorouting of wire `A -> MUX2` with the same `A1 -> B`, when moving `A` or `A1`. `A1` wire works as expected. `A` wire does not avoid the adder component.

**Expected behavior**

`A` routing should behave identical to `A1` routing

**Desktop (please complete the following information):**

 - Windows (but will not matter)
 - Version v4.0.a.12 (but anything > 4 will do this I think)

**Additional context**

It looks as though the MUX is causing different routing from other components, and creating the anomaly. Suggest trace through the SmartRouting function printing out what it does in the two cases, and compare.

theogam02, 2023-09-07:
Bug has now beed fixed. The problem was that the wire that goes into the select port of a mux or a demux always intersects with that mux's boundingbox, so the recursive autorouting reached maximum depth and aborted the autoroute effort.

closed 2023-09-07 by theogam02

tomcl, 2023-09-07:
That fixes my extracted anomaly. However, annoyingly, the original sheet I found it in is not fixed. Look at datapath in enclosed project. Maybe you can trace the alg on that and see why it still fails?

Tom