tomcl/issie #396: Code signing and notarisation is not working as expect

0x6770, 2023-09-08:
**Describe the bug**
The binary generated by `npm run dist` cannot be opened when download from GitHub release.

[image]

This is due to error on code signing and notarisation. 
1. `scripts/afterSignHook.js` expects wrong `appPath` for code signing. Naming format of Mac binary is changed.
2. `provisioningProfile: "embedded.provisionprofile"` is not found in the root directory.

tomcl, 2023-09-08:
:) Glad you have self-assigned!

0x6770, 2023-09-08:
Quick updates:

I'm still looking for a free way to code sign macOS app. It appears that membership of [Apple Developer Program](https://developer.apple.com/programs) is required to generate certificate and provisioning profile that are needed for code signing. (78 GBP for individual developer, free for nonprofit organization or educational institution, https://developer.apple.com/support/membership-fee-waiver/)

A safer temporary workaround is found that has no prerequisites and does not affect other apps. The `app is damaged` error #329 can be resolved by running

```
sudo xattr -rd com.apple.quarantine /Applications/issie.app
```

The installation path to issie may differ.

tomcl, 2023-09-12:
I am closing this now since Yujie's workaround seems to work and is in README.

closed 2023-09-12 by tomcl