tomcl/issie #397: Copied component is pasted with ('cmd' + anything)

theogam02, 2023-09-08:
**Describe the bug**
While working on the eep1 project, I copied components from one sheet to another. I pasted the components on the new sheet. After that I tried using some shortcuts to rotate the component. The moment I pressed 'cmd' (or control for windows), the component was pasted twice more. This seems to keep on going for quite a while. It is when I pressed 'cmd' + 'c' again to clear the clipboard that it stopped. Can someone try this on their computer (windows as well) and see if this happens. I am sure someone would have noticed by now if this happened to them but it seemed pretty weird to me. Perhaps it came with one of the latest commits?

**To Reproduce**
Steps to reproduce the behavior:
1. Use eep1 project
2. Copy component from Datapath to new sheet
3. Paste component
4. Try using other shourtcuts with ctrl/cmd (windows/mac)

**Expected behavior**
The component should only be pasted once, with the correct shortcut

**Desktop (please complete the following information):**
 - OS: MacOs
 - Version 13.2

tomcl, 2023-09-08:
It does not happen under windows - all is good.

You can trace all and then work out which messages the Cmd key press generates. on windows ctrl is used for additional functionality on custom components so we detect it - but it doe snot do anything else and is not confused with Ctrl + c.

tomcl, 2023-09-23:
> As an extension to the previous bug description:
>
> This bug stems from the fact that Cmd+C, Cmd+V are caught only by the ManualKeyDown case in the match statement. ManualKeyDown and ManualKeyUp implement a set which keeps track of the current pressed keys. CurrentKeyPresses is an attribute of the Sheet.model. Of course, this list is used to execute actions that require multiple keys to be pressed (such as shortcuts). When a shortcut is found in CurrentKeyPresses, sheet update generates a KeyPress Ctrl+C/V/A/W message (which is a sheet message). This works most of the time:
>
>   *   If the keys are at a normal pace one after the other, the system detects all the KeyDowns and KeyUps and takes the respective actions of adding or removing the keys from the set of currently pressed keys. This means that the system works properly and triggers the Cmd+C/V/A/W messages when detecting Cmd and C/V/A/W.
>   *   When the keys are pressed quicly one after the other, the KeyUp action sometimes is not detected and a key that is now unpressed remains in the register of pressed keys. This is how pasting happened by just pressing Cmd after Cmd+V had already been pressed. V was still in CurrentKeyPresses. Only if V was pressed again on its own, did it get removed from the set.
>   *   When the keys are pressed simultaniously, the behaviour was somehow better. I couldn't get it to go wrong.
>
> For anyone working on this bug in the future, I suggest starting by printing out the CurrentKeysPressed to get an understanding of what is described above.
>
> Potential solutions:
>
>   1.  Somehow make the scanning process for KeyUp, KeyDown faster.
>   2.  Find how electron can directly generate Keypress Ctrl+C/V/A/W messages for these shortcuts to avoid the ManualKeyDown way.

So the question is WHY is the keyup not detected. You can trace it through:
Message
Key down list.
Logic on that (maybe affected by sequence etc).

commit 9d2fa7d3 (2023-09-24): fix Ctrl-meta stuck down

Fixes #397 (I hope)
closed 2023-09-24 by tomcl