tomcl/issie #428: Installing Issie 4.1. on Macs

tomcl, 2023-10-30:
*NB* For best experience with Issie don't upgrade to Macos 13 (Ventura) unless you have to!

**Pre-Ventura apple silicon macs:**

arm64.dmg binaries as in readme (follow instructions to get round unsigned app security).

**Old (intel) macs:**

.dmg - this will also run on apple silicon macs under teh interpreter - but that is significantly slower than the am64 compiled version above

**Ventura macs:**
We are not yet how to get the arm64 binaries to work - it seems Ventura has changed security settings and previous documented fixes no longer work!

It is possible (please report experience if you have Ventura) that changing security settings will allow the arm64.dmg binaries to run. try googling Ventura fixes and let us know (reply this issue) if anything works.

it seems that .dmg binaries will still run ok (although be quite a lot slower than arm compiled binaries). So try installing the Issie4.1 .dmg binary as a temporary workaround.

Reply to this issue with any additional information (we do not have a Ventura Mac currently for testing).

jamesxsc, 2023-11-20:
I had no issues installing on MacOs 15 or 16. Run the following in terminal.

```
xattr -cr /path/to/issie.app
```

Source: https://osxdaily.com/2019/02/13/fix-app-damaged-cant-be-opened-trash-error-mac/

tomcl, 2023-11-20:
Just to be sure - is that with the arm64 binaries?

jamesxsc, 2023-11-20:
That was on 'issie-4.1.0-arm64.dmg'