tomcl/issie #429: Truth table using mergeNwire

jasyu12, 2023-11-05:
**Describe the bug**
 Wrong truth table results as using merge N bus wires to wire up the output

**To Reproduce**
Steps to reproduce the behavior:
1. Construct the deca section 1 part 2 ripple-carry-order, ensure the halfadd and fulladd merge block is working( as well as their truth table)
2. simulate riiple-carry-order using Step Simulation, the sum is correct
3. use truth table to generate the total output, some of the s (sum) result is incorrect
4. only mergeNwire will cause the problem, replacing with mergewire output goes well

**Expected behavior**
 
It is an add-up output, s will be the sum of A and B

[image]
0b0000 + 0b0100 = 0b0100

[image]
0b0000 + 0b1000 = 0b1000

**Screenshots**
 
[image]
at second line,  0b0000 + 0b1000 = 0b0010 third line 0b0000 + 0b0100 = 0b0010
many of them are invalid

tomcl, 2023-11-18:
This is the same as #304.

SplitN and mergeN are not needed - they can be replaced by split, merge or  bus select.

These components were added late in the Summer and alas the truth table simulation was not correctly implemented for them. This will have medium priority in the development. Apologies for the inconvenience.

tomcl, 2023-12-22:
Attempted fix as latest commit on master f583b490aef7ab57fc5e255824b47487ead330c7 and attached windows binary fix under v4.1.1

Also fixes #304 

These functions are quite complex and have been only partially tested. Feedback from use of the fixed binaries gratefully received.

tomcl, 2023-12-30:
Closing now since it seems to be fixed. Can be reopened if not!

closed 2023-12-30 by tomcl