tomcl/issie #430: Inability to import sheets from one project to another

Ivy-yu7, 2023-11-25:
**Describe the bug**
Using ISSIE software on my MacBook. The problem lies in the inability to import sheets from one project to another.

 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to any project. 
2. Click on top left Sheets - Import Sheet
3. Every sheet is grey (unable to be clicked on hence imported)

**Expected behavior**
Expect to be able to import sheets. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - macOS Monterey Version 12.3.1 (MacBook Air M1, 2020) Chip: Apple M1
 - I am using the latest version of ISSIE (v4.1.0)

**Additional context**
Add any other context about the problem here.
[image]

tomcl, 2023-11-25:
I think this is a misunderstanding of the import function.

To use it:

1. Open the project you wish to add a sheet to
2. Click import sheet from the sheets menu
3. Select the project directory, and sheet9s0 within the directory, that you wish to import
4. Follow any confirmation dialogs

tomcl, 2023-11-25:
This was in fact a bug in v4.1.0. Fixed now v4.1.1.

closed 2023-12-30 by tomcl