tomcl/issie #442: Long simulation bug

tomcl, 2024-03-23:
**Describe the bug**
White-screen crash with issie unresponsive after has been reported when doing long simulations of large designs.

**To Reproduce**
Not yet enough information

**Additional context**
This was reported Spring 2024 by an Issie user. It is hoped that they will be able to recreate the issue with dev tools open and report the stacktrace that will probably diagnose this problem.

tomcl, 2024-03-24:
It is _possible_ this is simply because currently the waveform simulator keeps SVG waveforms from 0 until the max time required. (I cannot remember what we now do).

Currently we have a supposed circular array which holds the last N simulation samples. Going forward from this in time is fine - the simulator restarts from the latest sample. Going back is problematic because the only way to go back is to resimulate from 0. The circular array is not used _properly_ as  a circular array but is fille dup so that the waveform simulator can be guaranteed to have a available the last MaxArraySize samples in the array.

The solution is to keep an array of records of checkpoint simulation state every 1000 clocks or whatever, and to make the circular array keep track of its starting sample as well as its last sample (making only the assumption that between these two times we have data correctly stored in the array). 

To add the additional data it will be necessary to add a field to `fc: FastComponent`:

```
HistoricState: StepData<SimulationComponentState> option // this is exactly like State field

(Option.get fc.HistoricState.StepData).Step[time] = 
     (Option.get fc.State.StepData).Step[time*fc.CheckPointInterval]
```
Where `CheckPointInterval` is a new filed of `FastComponent` which is initialised to be must be less than the circular array size used (`fc.MaxArraySize`). 

Also the "circular arrays" are not currently used properly as circular arrays. We would need a new mutable field `FirstClockTick` of `FastSimulation`.

Then at top-level when running a simulation: `fastSim:fastSimulation`  to clock `t` we would implement:

```
if t < fastSim.FirstClockTick then // in this case we have gone beyond circular array
    // find largest stored checkpoint < t  (= t / fastSim.CheckPointInterval
    // restart simulation at checkpoint time
// do whatever was done before to simulate.
```

to restart a simulation at some checkpoint time:

* Set `FirstClockTick `to the relevant checkpoint time
* Set `ClockTick `= `FirstClockTick` (or possibly `FirstClockTick - 1`)
* For each `FastComponent` with state, set its State at array index (`FirstClockTick/ MaxarraySize`) equal to the corresponding stored `HistoricState` containing the historic state data

When running the simulation:
* set `FirstClockTick `to be min 0 (`ClockTick-MaxArraySize`)

Then resimulating any time that has been previously simulated requires at most 1000 cycles of simulation.

When making this (overdue, really) change to the simulation to handle long simulations properly it would also be possible to use HistoricState (if it exists) to go forward in time. The procedure would restart the simulation and some future checkpoint time (the largest less than the time required) and be identical to when going back in time.

Adding this to the simulator would be pretty easy, and would actually make the simulation have a more uniform user interface where at top level you simply ask for a time range of simulated data (containing no more than MaxArraySize steps) and the simulator will do the work necessary to give you that.

samuelpswang, 2024-03-25:
**Reproduction of Bug Failed**

* Simulated all signals in demo EEP1 circuit, going cycle-by-cycle or jumping to specific cycles seems to work fine.
* Cycles simulated seems to be capped at 1000, not sure whether this is intentional, will investigate further later.

tomcl, 2024-03-28:
Well done! It is a bit of a mystery to me. 

One suggestion - it is a memory problem. perhaps keeping dev tools open and  graphing memory use and recreating the bug would show this (not sure whether dev tools can provide continuous indication of memory use - rather than graphing it at end after which maybe it will have crashed?).

Another suggestion: trace SVG generation. That would seem to be the issue?

Finally - there is some code that puts up a progress control when SVG generation takes too long. Maybe that is going wrong - but the issue I do not understand is why no backtrace. Perhaps in some circumstances an infinite loop?

tomcl, 2024-03-28:
The heap size goes up above 500MB in this situation and given dev tools itself uses heap it looks like the heap limit may be the problem.

https://medium.com/weekly-webtips/increase-node-js-memory-limit-bonus-pm2-62cda081ce2a#:~:text=js%20(up%20to%2011.,js%20application.

The bug happens when expanding a small window for up to 1000 clocks.

https://github.com/electron/electron/issues/2056

Could be followed to increase the heap size and see if that helps - but we probably also want to limit the number of waveforms we create

tomcl, 2024-03-28:
I have tried implementing that fix. See commit on `dev_tc` for what I have done. I am not even sure the heap size is being increased because I cannot easily get the v8.getHeapSatatistics() method to work - it causes an infinite loop. _Possibly_ this requires "require `v8`" -  all this js stuff is a bit opaque to me, and because it is js via F# and fable it is obscure to a lot of people.

I am still not convinced this is a heap limit. Printing out memory usage after each simulation is crashed at 720MB. That is suspiciously close to the standard 32 bit memory limit of 700MB or so. But surely we run a 64 bit version of electron? And the fix to increase hep size to 7500MB should be working...

I guess it needs checking:

* is electron 32 bit?
* What is the actual heap size limit (could run a dedicated infinite memory hog loop under Issie and check help usage until crash - that would make it clear do we really have a heap problem, or something else

tomcl, 2024-03-29:
More info (see latest commit on dev-tc)

* a test program using heap can get up to measured 4200MB heap before crashing.
* the fix above is not needed, and makes no difference to this memory limit
* Adding print statements: the carsh occurs OUTSIDE of any of Issie's code after the view function has terminated when the react generated by the view is given to electron chromium to render.

It looks to me like:

* possibly a cyclic bit of react - difficult to see how that can be generated
* likely a malformed bit of svg which usually is ok but sometimes causes crashes in renderer
* possibly some overflow in the react svg renderer caused by the many very large svgs generated when the svgs are 1000 cycles long
* possibly some bug in the react svg renderer.

What we get from this:

1. We should implement the heap expansion - at least to 3500MB (will run in 4GB machines) - and optimally to 7500MB. It will speed up Issie!
2. This bug looks tough to find but as a workaround we should just limit the number of waveforms displayed to say 50.
3. to limit the number of waveforms displayed nicely we need a modified wave select UI
4. We should not allow more than (say) 250 cycles displayed on the screen - or if more are displayed use a modified display that reduces the number of svg points, e.g. display oscillating waveforms as a grey rectangle.

tomcl, 2024-03-29:
The main process crashes giving the following backtrace:

FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory
 1: 00007FF6AB500D36 node::SetTracingController+80134
 2: 00007FF6AB500FF5 node::OnFatalError+645
 3: 00007FF6AE15B873 v8::Function::NewInstance+851
 4: 00007FF6AE15B808 v8::Function::NewInstance+744
 5: 00007FF6AE21C417 v8::CppHeap::CollectGarbageInYoungGenerationForTesting+67671
 6: 00007FF6ACDFF0E6 v8::CppHeap::GetHeapHandle+86326
 7: 00007FF6ACDFA526 v8::CppHeap::GetHeapHandle+66934
 8: 00007FF6ACDEC626 v8::CppHeap::GetHeapHandle+9846
 9: 00007FF6AD012613 v8::CppHeap::GetHeapHandle+2262627
10: 00007FF63FE9C3F9
Error sending from webFrameMain:  Error: Render frame was disposed before WebFrameMain could be accessed
    at s.send (node:electron/js2c/browser_init:2:86493)
    at _.send (node:electron/js2c/browser_init:2:72283)
    at BrowserWindow.<anonymous> (C:\GitHub\issie\build\index.js:21626:25)
    at BrowserWindow.emit (node:events:525:35)
Error sending from webFrameMain:  Error: Render frame was disposed before WebFrameMain could be accessed
    at s.send (node:electron/js2c/browser_init:2:86493)
    at _.send (node:electron/js2c/browser_init:2:72283)
    at BrowserWindow.tupledArg (C:\GitHub\issie\build\index.js:21631:27)
    at BrowserWindow.emit (node:events:513:28)
Error sending from webFrameMain:  Error: Render frame was disposed before WebFrameMain could be accessed
    at s.send (node:electron/js2c/browser_init:2:86493)
    at _.send (node:electron/js2c/browser_init:2:72283)
    at BrowserWindow.<anonymous> (C:\GitHub\issie\build\index.js:21626:25)
    at BrowserWindow.emit (node:events:525:35)

tomcl, 2024-03-29:
To wrap up the **heap size maze**.

https://github.com/electron/electron/issues/31330

**It turns out that a hard heap size is limited to 4GB by a v8 pointer compression feature that electron will not ever be able to switch off. Total memory size is not so limited - unfortunately in F# almost everything is allocated on heap.**

https://www.electronjs.org/blog/v8-memory-cage

That is sad - but not disastrous since Issie does not use anything like that amount of memory. The heap out of memory message here must be related to the SVG rendering process.

The symptoms are that this error happens only when making a large number of long SVGs. 

* It is possible that the total heap size used by the SVG renderer is exceeded, because of too many SVG objects with too many points on them.
* It is possible this is a bug in the SVG / react rendering code
* It is sort-of possible that some malformed svg triggers a bug that has the effect of increasing heap usage.
* It is still unclear which process this heap size overflow comes from - I guess some react process that actually generates the screen pixels independent of the renderer? Because the renderer heap size does not get so large.

Given this, my best guess is that the problem is the total SVG polyline path size, or the total SVG font size. We have 4 path points and up to one text element with 10 chars per clock cycle. Currently 1000 clock cycles are always generated.

It is therefore plausible that even though most of this stuff is not displayed, the SVG object size is too large. Part of the polylines is displayed and it is possible that therefore they are all processed. For 200 waveforms we have 1,000,000 polyline points.

**Fixes**
* Quick and dirty - limit number of waveforms. This probably _should_ be limited, because with 100 waveforms it is difficult to find the one you want. 
* Correct. Generate *only* the visible portion of SVG. Do this dynamically on horizontal software scrolling. Create SVG text for all waveforms - but place this on react SVG only for the visible waveforms based on vertical scrolling: delete from react SVGs not currently visible. 
* Limiting *selected* SVG number is now not much needed - but should even so limit this to say 100.

tomcl, 2024-03-30:
checking the conditions for this crash to happen:

1. it never happens when waveforms are zoomed out
2. it happens when waveforms are zoomed in
3. The heap use increases a large amount as waveforms are zoomed in (before the crash happens)

[image]

This shows memory use as waveforms are remade under conditions that will cause the crash at high zoom-in. The three levels of memory use correspond to making waveforms after pressing the zoom-in button twice. Each time memory usage goes up, and also the time taken to make the waveforms increases by a large amount.

Memory allocation trace shows most of the allocated memory is SVG text:
higher zoom allocations:
[image]

lower zoom allocations:
[image]

samuelpswang, 2024-08-05:
As suggested by @tomcl, the white screen crash was mainly due to large memory usage. Improving the SVG generation as #448 in PR #456 was sufficient to resolve this issue.

closed 2024-08-05 by samuelpswang

tomcl, 2024-08-16:
Just a comment - this bug was largely due to the Issie memory leak now fixed.
See #463