tomcl/issie #446: Can't Run Issie

Letmein456, 2024-03-28:
Whenever I download the issie 4.1.3 zip, unzip, and then open it, I can't find the issie.exe file. I don't know if I should be running another file or what?

- OS Windows 10
- Version 4.1.3

samuelpswang, 2024-03-29:
Hi @Letmein456, just to confirm, are you running unzipping this [`issie-4.1.3-win.zip`](https://github.com/tomcl/issie/releases/download/v4.1.3/issie-4.1.3-win.zip) from the releases page?

(I downloaded it and unzipped it, but `issie.exe` seems to be there.)

Letmein456, 2024-03-29:
I didn't see where to download it from the releases page so I went to the actual file in GitHub, clicked code, and clicked download as zip

Letmein456, 2024-03-29:
I think my issue is that I'm downloading in the wrong area, can you detail to me how to download from the releases page?

samuelpswang, 2024-03-29:
If you go to the main page ([https://github.com/tomcl/issie](https://github.com/tomcl/issie)) of the repo, and you will see the Releases tab on the right. If not, you can try the release page URL directly ([https://github.com/tomcl/issie/releases](https://github.com/tomcl/issie/releases)). You will be able to find the versioned releases there.

Please see the [Getting Started](https://github.com/tomcl/issie?tab=readme-ov-file#getting-started) section of the main README file.

closed 2024-03-29 by samuelpswang