tomcl/issie #484: Multi-page Truth Table Navigation

samuelpswang, 2024-10-25:
**Is your feature request related to a problem? Please describe.**
The Truth Table feature under the Simulation tab does not have an intuitive way of seeing all rows of the displayed truth table. Students reportedly asked questions on how to see all rows.

**Describe the solution you'd like**
A scrollbar, either in HTML or via SVG canvas, would greatly help.

**Describe alternatives you've considered**
Current alternative is to use up-down buttons, which is not obvious to students.

tomcl, 2024-11-20:
Truth tables are sort of an optional add-on for Issie. But I guess they now need some TLC. This is noted for possible dev in the Summer (sorry everyone this year). Thanks.

Re seeing all rows: partial solution: use Ctrl-Shift -  to zoom out.

samuelpswang, 2024-12-12:
See discussion in #486, fixed in #490 (v5.2.2).

closed 2024-12-12 by samuelpswang