tomcl/issie #486: Unable to see bottom part of the truth table - Scrolling not working

Emiral46, 2024-10-29:
**Describe the bug**
In truth tables part of the program, when I generate a large truth table with a lot of possible combinations, only the top part of the table is visible in the page and I cannot scroll through the truth table, preventing me from seeing all the outputs and inputs in the table
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Truth table in the right plane while having more than 4 input gates'
2. Click on 'Generate truth table'
3. Try to scroll through the table to see all the possible combinations of the inputs and outputs
4. See that you won't be able to scroll through it.

**Expected behavior**
I should be able to see all the inputs and outputs of the truth table.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: MacOS but same also happens on my friends computer running Windows 11.
 - Version v5.2.1

**Additional context**
I am a first year EIE student at Imperial and I am not able to do my first year lab because of this issue.

tomcl, 2024-11-20:
Ah...

Maybe this is a bug?
Workaround - use Ctrl-shift - to scroll out.

I will check if the truth-stables were ever meant to scroll, if so we can probably make it work easily.

tomcl, 2024-12-01:
Fixed by #490 v5.2.2

I've generated windows binaries. Currently only @samuelpswang can generate signed versions of the Macos binaries - due to new Macos security restrictions the unsigned ones I can generate are not very usable. We will sort this out shortly, till then unsigned Macos binaries can be generated (it is quite easy - see README) on the Mac they are used on.

samuelpswang, 2024-12-10:
Quick note: `v5.2.2` binaries for macOS has been generated and signed, and posted to the Releases page; Linux binaries will follow shortly.

samuelpswang, 2024-12-12:
Update: Linux binaries compiled and uploaded to Releases page.