tomcl/issie #489: Unable to view the complete truth table in the truth table generator

LH-and-FPGA, 2024-11-30:
**Describe the bug**  
When using the software for simulation and generating a truth table, if the truth table is very large and exceeds the screen size, I cannot scroll down to view the complete truth table. Additionally, there is no option to export the truth table as an image.

**To Reproduce**  
Steps to reproduce the behavior:  
1. Open the simulation tool in the software.  
2. Generate a large truth table that exceeds the screen height.  
3. Attempt to scroll down to view the full table.  
4. Notice that scrolling is not possible, and there is no export option available.

**Expected behavior**  
I expected to be able to scroll down and view the entire truth table or have an option to export the table as an image for external viewing.

**Desktop (please complete the following information):**  
- OS: Windows 11

**Additional context**  
The lack of scrolling and export functionality makes it difficult to analyze or save large truth tables generated during simulations. Adding these features would significantly improve the user experience.

tomcl, 2024-11-30:
Thanks for that. Closed in https://github.com/tomcl/issie/releases/tag/v5.2.2

tomcl, 2024-11-30:
One comment: this is a useful fix but rather than go down the route of very large truth tables it would be better to enhance the algebraic truth table logic. The following are possible:

* allow better simplification rules (this however is quite complex)
* allow more inputs to be algebraic (currently a few components do not allow algebraic propagation when they could)
* Make the algebra button default to having as algebraic all inputs allowed to be so. That is nearly always what is wanted.

LH-and-FPGA, 2024-11-30:
Thank you very much for your comments and for merging my code. However, regarding the second point, since I am still in my EEE first year, I am quite curious about which components do not allow algebraic propagation? 

Moreover, I am working on the third one. I was wondering if it might be worth considering adding a `canBeAlgebraic` function check before the `truthTableUpdate` function. This function could attempt to generate the Truth Table but stop if it detects an excessively complex expression (For example, in our First Year lab, if the select line `S` of a MUX is set to algebraic, it can result in extremely complex expressions). But the question here is how to define "complex" in general.

closed 2024-12-26 by LH-and-FPGA