tomcl/issie #505: XOR Logic Display Issue in Issie v5.3.0 (Fixed in PR #501)

LH-and-FPGA, 2025-02-26:
**Describe the bug**  
In Issie v5.3.0, the XOR logic does not display correctly (due to the logic simplification). However, this issue was resolved in my previous PR #501, which has already been merged.  

**To Reproduce**  
Steps to reproduce the behavior:  
1. Open Issie v5.3.0.  
2. Add an XOR gate to the circuit.  
3. Observe the "incorrect" logic display.  

**Expected behavior**  
The XOR gate should display and function correctly, as expected based on the fix implemented in PR #501.  

**Screenshots**  
New Version:
[image]
Current Version:
[image]

**Desktop (please complete the following information):**  
- **OS**: Windows 11
- **Version**: Issie v5.3.0  

**Additional context**  
Since the issue has already been fixed in PR #501, could you please publish a new version to reflect this fix? Thank you!

tomcl, 2025-02-26:
I have redone the Windows binaries  release: without changing version number. If i do that everyone will feel they have to download a new version.

It just needs somone to make macos and (eventually) linux.

If the buggy feature is actively being used now I can make a new version.

tomcl, 2025-02-26:
closed by https://github.com/tomcl/issie/releases/tag/v5.3.1

closed 2025-02-26 by tomcl