tomcl/issie #508: documentation build does not work

tomcl, 2025-03-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, MacOs]
 - Version [e.g. v3.0.0]

**Additional context**
Add any other context about the problem here.

tomcl, 2025-03-08:
Scheduled for attention Summer or possibly Spring 2025

samuelpswang, 2025-09-02:
Fixed in #562: Token required by `peaceiris/actions-gh-pages@v4` refers to the [GitHub Workflow's token](https://docs.github.com/en/actions/concepts/security/github_token), and _not_ the Personal Access Token of the owner of the repo. See further docs [here](https://github.com/peaceiris/actions-gh-pages?tab=readme-ov-file#%EF%B8%8F-set-runners-access-token-github_token).

Broken in #563: Typo in `build_docs.yml` on L11. To be fixed in next PR.

samuelpswang, 2025-09-04:
Do you mean the Test Suite workflow that has 19 or so tests on several different OSes? I think those are introduced into the codebase by LH-and-FPGA in [1bc5bee](https://github.com/tomcl/issie/commit/1bc5beed8520ef315af59106a46a0e8962cc2a75#diff-faff1af3d8ff408964a57b2e475f69a6b7c7b71c9978cccc8f471798caac2c88).

tomcl, 2025-09-04:
Ah - yes. that makes sense. I will point out to Han that his tests should pass.