tomcl/issie #526: Adding a rename keyboard shortcut

SanjitRaman, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
- it would be useful to have a keyboard shortcut to rename a component.

**Describe the solution you'd like**
Click on a component, hit F2 and type in the new name.

**Describe alternatives you've considered**
- any other keyboard shortcut would also be nice.

SanjitRaman, 2025-04-04:
That seems fair, however, F2 is pretty common across different software so as a user I kind of expected it to work.

SanjitRaman, 2025-04-04:
actually, I just noticed that for wire labels, it doesn't open up the name by default

tomcl, 2025-04-04:
OK. F2 common for renaming? We do not currently using functions keys as shortcuts. I am happy to use them if they will work reliably cross-platform? In which case we could have that as a shortcut.

I suspect there are many other convenience UI features on Issie that are higher priority - but we do not always think of them. So please also post here anything else. I will keep this issue open for discussion with the idea that at some point in the Summer we review shortcuts.

PS - the worst part of the Issie UI - the waveform simulator wave selector - has now been redone thanks to this year's 3rd year HLP cohort. It is (I think) great. And those annoying greyed out parts of waveforms now have nice tooltips. Try v5.9.3

SanjitRaman, 2025-04-04:
ok, so I just finished making 64-bit register bank, and the only sane way to do it was with some python scripting. I nosied into the format of DGM files, and ended up writing some scripts:
- autoconnect_by_index_matching.py
- autoconnect_by_y.py
- rename.py for labels
- sort_by_y.py for renaming components into numerical order.

I suppose some cleaned-up/generalisable version of these features would be very useful.

SanjitRaman, 2025-04-04:
with regards to the F2 shortcut, I think I noticed that sometimes the properties tab doesn't automatically show up when you copy-paste a component. This is still on v5.3.1

SanjitRaman, 2025-04-04:
another user thing I noticed on v5.3.1:
- when you press sheets and then click on the main schematic area, I think I would expect the menu to disappear on an "off-click".

SanjitRaman, 2025-04-04:
shortcut request
- upon filling out the parameters e.g. input name and width, ENTER should place the component. Right now, enter just disappears the data one has just entered.

LH-and-FPGA, 2025-07-24:
Currently, issie has following keyboard shortcuts, I want to make sure there are no further change needed before I start:
 # Essential Shortcuts

  ## File & Edit Operations:
  - Save: Ctrl+S (Win/Linux) / Cmd+S (Mac)
  - Copy: Ctrl+C / Cmd+C
  - Paste: Ctrl+V / Cmd+V
  - Select All: Ctrl+A / Cmd+A
  - Undo: Ctrl+Z / Cmd+Z
  - Redo: Ctrl+Y / Cmd+Y
  - Delete: Delete key
  - Cancel: Escape

  ## View Control:
  - Fit to Window: Ctrl+W / Cmd+W
  - Zoom In: Alt+Up or Ctrl+Mouse Wheel Up
  - Zoom Out: Alt+Down or Ctrl+Mouse Wheel Down

  ## Component Manipulation:
  - Rotate Clockwise: Ctrl+Right Arrow
  - Rotate Counter-clockwise: Ctrl+Left Arrow
  - Flip Vertical: Ctrl+Up Arrow
  - Flip Horizontal: Ctrl+Down Arrow
  - Move Ports: Ctrl+Drag (on custom components)

 #  Special Modes

  ## Modifier Keys:
  - Ctrl/Cmd Hold: Enables port movement and multi-selection
  - Shift Hold: Enables canvas panning and additive selection
  - Space: Disabled to prevent page scrolling

  ## Context-Specific:
  - WaveSim: Number keys (0-9) change multiplier, Arrow keys navigate
  - Code Editor: Has its own key handling when active

  # Platform Notes

  - Windows/Linux use Ctrl, macOS uses Cmd (Meta)
  - Some shortcuts have manual implementations to ensure they work across      
  platforms
  - Browser zoom (Ctrl+Shift+Plus/Minus) is discouraged - use Issie's
  built-in zoom instead

tomcl, 2025-07-24:
Looks ok to me - but you may have more info about what people expect.

Known Issue: when properties are open in design sheet we use keys to control editing in property boxes. There is some lack of clarity over how we process keys and when they go to properties boxes, when they are used to navigate the schematic. Current solution to this is a bit hacky. Getting this wrong leads to **surprising** behaviour where the schematic sheet keys seem not to work - because they are being turned into input box operations. This is undesirable

tomcl, 2026-09-07:
This is now fixed via the v5/v6 Ui chnaged this summer. The keys are now standardised, with Shift-drag moving canvas. There have ben substantial changes to focus behavior in db/properties etc. If it is still not right this can be reopened.

closed 2026-09-07 by tomcl