tomcl/issie #528: Sort duplicated components by default.

SanjitRaman, 2025-04-04:
**Describe the bug**
I want to create a bunch of registers for a register bank for example. I create 2 and then duplicate to 4, 8 16, 32. When you duplicate the registers, they are named in a random order (vertically down the screen). So one has to go about renaming them individually which doesn't save any time.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
when you paste many of the same element, they are according to their vertical (or horizontal position).
- perhaps the best would be to allow "Bulk add components?" which does this by default.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, MacOs] linux
 - Version [e.g. v3.0.0] 5.3.1

**Additional context**
Add any other context about the problem here.

tomcl, 2025-04-04:
The long-term solution to this is better ways to generate arrays of components. That is on the roadmap for eitehr this Summer or more likely next year.

For a short-term fix in general we cannot sort duplicated components because they have positions that are related to routing and normally need to be preserved. The best solution would be something like "align" that took a selected group of components and sorted them lexicographically by label. Would that work?

For your use case you could also move them by hand into the  right order (v quick) and then do align and distribute to gte poistions eaxct?

SanjitRaman, 2025-04-04:
Looking forward to arrays!

in the meantime,
> The best solution would be something like "align" that took a selected group of components and sorted them lexicographically by label. Would that work?

yes, that would be so great. moving by hand is quite tedious because the problem scales.